weisongwen/UrbanNavDataset #5: What does the topic "/navsat/odom" mean?

Tompson11, 2020-08-08:
Hi:
Thank you for sharing the datasets!
I wonder if the topic "/navsat/odom" is the groundtruth transformd to Cartesian coordinate from SPAN-CPT coordinate?

DarrenWong, 2020-08-08:
topic "/navsat/odom" is the UTM coordinate based on the original point (Where we've started) of  topic "/navsat/origin", you can find more details from [here](https://github.com/ros-drivers/novatel_span_driver/blob/513f7059a38ed8a5f8850eecc75d9d771b51de50/novatel_span_driver/src/novatel_span_driver/publisher.py)

We recommend to use "/novatel_data/inspvax" for the ground-truth or download it directly by [dropbox link](https://www.dropbox.com/s/g1xt89py1j0uea1/IMU-Reference%20UrbanNav-HK-Data20190428.zip?dl=0) or BaiduCloud link in the readme

closed 2020-08-10 by DarrenWong

Tompson11, 2020-08-15:
Thank you for your help!