weisongwen/UrbanNavDataset #6: Methods and Tools to Calibrate Extrinsic

zxl19, 2020-08-20:
Hi @DarrenWong @weisongwen !

Thank you for opening this dataset! I have a question regarding the calibration methods. I'm using my own data and also need the extrinsic among IMU, LiDAR, GNSS and groundtruth(from OxTS). I wonder what methods you used to calibrate those extrinsic in your dataset. I really appreciate it if you could list some papers about the calibration methods you employed. If you have used open-sourced tools for calibration, it would be so much the better to provide names or links. Thanks in advance.

Best wishes.

DarrenWong, 2020-08-20:
For Lidar-IMU extrinsic calibration we use the least-squares method.  GNSS related are calibrated based on the CAD model. 
Might be you can find some useful information from https://github.com/APRIL-ZJU/lidar_IMU_calib

zxl19, 2020-08-20:
Thank you for your reply! @DarrenWong 

I would really appreciate it if you could specify **the least-squares method** you mentioned a little bit. Is it some sort of hand-eye calibration? I've also followed the paper you mentioned before, but for car-bourne IMUs it's really hard to stimulate the IMU in the vertical direction so that the result may not be that accurate. How did you solve this? I'm new to SLAM and any advice would be helpful.

Best regards.

DarrenWong, 2020-08-21:
We use the least-squares approach to solve the nonlinear optimization by 6D motion alignment from different sensors.
To my knowledge, hand-eye calibration mainly used together with the robot arm, for the robot problem of the forms AX = XB and AX = YB is given.

closed 2020-08-21 by zxl19