weisongwen/UrbanNavDataset #8: Cannot load message class for /novatel_data/inspvax.

zxl19, 2020-10-18:
Hi @weisongwen @DarrenWong !

Thank you for the data provided! I tried to record a clip of the ground truth topic `/novatel_data/inspvax `, but I didn't get it in the resulting bag. I tried to print the topic, and get the following error:

```shell
rostopic echo /novatel_data/inspvax 
ERROR: Cannot load message class for [novatel_msgs/INSPVAX]. Are your messages built?
```
I read the documents of [novatel_span_driver](http://wiki.ros.org/novatel_span_driver), but didn't find the answer. I wonder what caused the problem.
Also I read [issue#5](https://github.com/weisongwen/UrbanNavDataset/issues/5#issue-675392455), can I use `/navsat/odom` combined with `/navsat/origin` as the ground truth? If so, where is the origin point located? Is it the IMU coordinate system in the first frame?

Best wishes.

DarrenWong, 2020-10-18:
Do you install the package of novatel_msg?
```
sudo apt install ros-kinetic-novatel-msgs  # for ubuntu 16.04 with ros kinetic
sudo apt install ros-melodic-novatel-msgs  # for ubuntu 18.04 with ros melodic
```

zxl19, 2020-10-18:
Hi @DarrenWong !
Thank you, it's working now!

Also, can I use /navsat/odom combined with /navsat/origin as the ground truth? If so, where is the origin point located? Is it the IMU coordinate system in the first frame? I'm still confused right now.

zxl19, 2020-10-18:
I managed to save the `/novatel_data/inspvax` topic, but it seems that there is no timestamp in the message structure. How can I save the message with `.header.stamp.secs` and `.header.stamp.nsecs`?

DarrenWong, 2020-10-19:
> Hi @DarrenWong !
> Thank you, it's working now!
> 
> Also, can I use /navsat/odom combined with /navsat/origin as the ground truth? If so, where is the origin point located? Is it the IMU coordinate system in the first frame? I'm still confused right now.

You can use the first data of  lat, lon and azimuth in  `/novatel_data/inspvax` to represent the origin point approximately. But you need to align it with your IMU coordinate using the relative translation and rotation . The extrinsic of  UrbanNav-HK-Data20190428 is here https://github.com/weisongwen/UrbanNavDataset/blob/master/UrbanNav-HK-Data20190428/extrinsic.yaml

> I managed to save the `/novatel_data/inspvax` topic, but it seems that there is no timestamp in the message structure. How can I save the message with `.header.stamp.secs` and `.header.stamp.nsecs`?

The `/novatel_data/inspvax` topic save the GPS timestamp, you need to convert it to UTC time then align with ROS time if you want to compare the timestamp ( it might have a small stable time offset to ROS time)
A simple GPS-UTC convert for reference https://www.andrews.edu/~tzs/timeconv/timeconvert.php

closed 2020-10-29 by DarrenWong

Liming-Cheng, 2020-12-03:
@zxl19 
Hi, do you get the ground_truth with timestamp ?

DarrenWong, 2022-12-07:
The Novatel msg might not support the Neotic from [here](https://github.com/swri-robotics/novatel_gps_driver#usage). Suggest you can download the package and place it under src folder, then build it from source