weisongwen/UrbanNavDataset #12: The difference between coordinates

BayRanger, 2021-07-07:
Hey there,
I just ran the lio-sam with the second hk dataset. It works fine, But from the  observation. the map coordinate cannot coincide with the base link even at the beginning with an obvious yaw difference
The reason why I concerned about this problem  is because it would lead to inconvenience when comparing with the ground truth data.    
So would you give some hints about how to make the coordinates coincide so as to compare the odom output with ground truth?
[image]
[image]

DarrenWong, 2021-07-07:
LIO-SAM will rotate the yaw based on the IMU orientation for better alignment with GPS data. If you don't use GPS data,  try to revise the 
https://github.com/TixiaoShan/LIO-SAM/blob/3a3d82fac4ba614ff2324be4ef83381b4256cc48/include/utility.h#L263
to

   ```
 // rotate roll pitch yaw
        Eigen::Quaterniond q_from(1,0, 0, 0);

        //Eigen::Quaterniond q_from(imu_in.orientation.w, imu_in.orientation.x, imu_in.orientation.y, imu_in.orientation.z);
```

BayRanger, 2021-07-07:
Great! Thanks for you hint, it works well!

closed 2021-07-07 by BayRanger