weisongwen/UrbanNavDataset #13: Ground truth of UrbanNav-HK-Data20200314

zjjdes, 2021-07-16:
Hi @DarrenWong @weisongwen,

Thanks for providing this dataset!

I'm using `UrbanNav-HK-Data20200314`, mainly the ground truth, UBLOX GNSS and LiDAR data. I have the following questions:

1. In the extrinsic parameters the transformations between IMU and LiDAR and between IMU and SPAN-CPT are both identity matrices, are these correct? If so, does this mean that the LiDAR, Novatel and IMU data in the rosbag are already aligned?
2. Is there a transformation between UBLOX and SPAN-CPT data?
3. Is there any information of the base station used by SPAN-CPT? I tried looking for REFSTATION messages in the rosbag but couldn't find any. The reason why I'm asking this is that when I compute the positions using the RTK technique (double-differenced carrier phase) using UBLOX data and HKQT as the reference station, I get a trajectory that seems to have an offset from the SPAN-CPT ground truth:
[image]
4. I have also looked at your `UrbanLoco` dataset in California. It seems that I can only find GPS and GLONASS from UBLOX. Does it have any route with other GNSS such as Gelileo or Beidou?
5. Is there any, or will you publish more datasets with dual-frequency raw GNSS observations?

Thank you for your work and help!

DarrenWong, 2021-07-17:
Q1. Yes, you are correct for the data UrbanNav-HK-Data20200314. In UrbanNav-HK-Data20190428 we have a rotation of 90 degrees between the LiDAR-to-IMU

Q2.  We do not have the transformation between UBLOX and SPAN-CPT currently. Normally we place the antenna around 50 cm front or rear part of the vehicle roof. 

Q3. The message from the station for SPAN-CPT RTK calculation is transmitted by the internet in a real-time manner. We do not save the measurements from the reference station. Regarding the reference station, we suggest the HKSC station which is closer to our test area.

Q4. NO. Only GPS and GLONASS satellites being included in the California dataset. GPS/BeiDou for HK dataset.

Q5: We will update some new datasets in UrbanNav in the coming months. Just keep an eye on the update :-)

zjjdes, 2021-07-17:
> Q1. Yes, you are correct for the data UrbanNav-HK-Data20200314. In UrbanNav-HK-Data20190428 we have a rotation of 90 degrees between the LiDAR-to-IMU
>
> Q2. We do not have the transformation between [...]

Thank you so much, that clears a lot!

closed 2021-07-17 by zjjdes