weisongwen/UrbanNavDataset #15: What does Roll, Pitch, Heading mean? 数据集中的Roll, Pitch, Heading表示的是什么意思呢？

Rotatingpencil, 2021-09-28:
Hello! Thank you for the excellent work!

Is it true that those data in references.csv file represent the rotation from SPAN to ENU coordinate of current frame? 

您好，感谢您提供的数据集！
真值文件中的角度信息代表的含义是什么呢？是代表从SPAN旋转至当前帧东北天坐标系的旋转信息吗？

Rotatingpencil, 2021-10-02:
@DarrenWong @weisongwen Hello to interrupt you again, it's really important for me.

DarrenWong, 2021-10-03:
Sorry for the late reply. The roll, pitch yaw is based on the Body Frame defined by Novatel. Roll is about the y-axis (Right-handed rotation), the pitch is about the x-axis (Right-handed rotation)., and azimuth/yaw is about the z-axis (Left-handed rotation). In UrbanNavDataset, Y pointing in the direction of travel, X to the right (perpendicular to the direction of travel), and Z up. 

You can change the data from body frame to ENU following below info:
Relating the Body Frame to the Local Level Frame, https://hexagondownloads.blob.core.windows.net/public/Novatel/assets/Documents/Bulletins/apn037/apn037.pdf

Rotatingpencil, 2021-10-05:
Thanks a lot! It's really helpful!