weisongwen/UrbanNavDataset #21: Using Higher Channel LiDAR

robertsenputras, 2022-05-18:
Hi weisongwen,

I’m so glad that I found this repo. The dataset which is specific in an urban canyon environment can provide a challenge to the developer. I plan to use this in the future. However, I found that the lidar that used in this dataset is only a 32 Channel Velodyne 32E LIDAR. I also arrived at your lidar super resolution paper with only using LiDAR data to get higher LiDAR channel such as 64 channel lidar. 

I’m wondering that is it possible for me to use your lidar super resolution lidar to 64 channel of this dataset. If you don’t mind, perhaps you can share the lidar enhancement algorithm with me just for benchmarking purposes?

I hope hearing from you soon. Thanks

weisongwen, 2022-05-30:
Hi @DarrenWong , can you please kindly help with this?

weisongwen, 2022-06-02:
Hi @robertsenputras, Thank you for your interest in our dataset. I think I mistake in your question. 

I think you are talking about this paper (Yue, J., Wen, W., Han, J. and Hsu, L.T., 2021. 3D Point Clouds Data Super Resolution-Aided LiDAR Odometry for Vehicular Positioning in Urban Canyons. IEEE Transactions on Vehicular Technology, 70(5), pp.4098-4112.). Unfortunately, this algorithm is from an industry collaboration project which currently cannot be easily open-sourced.

robertsenputras, 2022-06-15:
Hi @weisongwen . 

Thanks for your response. I can understand well that the algorithm cannot be easily open-sourced. However, I'm very interested with your research about the method that you use to combine two different output from two different models (in this case the sparse cnn and erfnet). However, I noticed that erfnet will take 3 input channels (rgb). How did your team convert the array of depth value (range image) which has only 1 channel in order to be used for erfnet which requires 3 channels? If you don't mind, maybe you can give a short explanation about it. 

Thank you.

DarrenWong, 2023-01-25:
closed as no further updates

closed 2023-01-25 by DarrenWong