weisongwen/UrbanNavDataset #22: About ground truth

zhanglei1997, 2022-05-31:
Hello, I also noticed another paper of yours:
Point-wise or Feature-wise? Benchmark Comparison of Public Available LiDAR Odometry Algorithms in Urban Canyons.
For the experimental part of this paper, you used two sequences that were previously made in Hong Kong. I would like to know how you used the EVO tool to get the ground truth from the bag and compare it with all the other algorithm results.
Looking forward to your reply.

zhanglei1997, 2022-05-31:
[image]

DarrenWong, 2022-06-02:
closed as same issue here: https://github.com/IPNL-POLYU/UrbanNavDataset/issues/18

closed 2022-06-02 by DarrenWong