weisongwen/UrbanNavDataset #23: Where's the data of fisheye camera?

NAI-ZHU, 2023-01-18:
Hi  weisongwen,

I'm so exacting to find this amazing work. But I have a little problem with the setup of the camera.
I noticed that, from the picture in 1.1 Sensors Setup, there is a sky-facing fisheye camera on the top of the car. But I only found the 1920*1200 image data of a foward-facing camera in the ROS dataset. Is there actually only one camera on the car or did I just miss the fisheye camera data? 

I'm looking forward to you reply. Thank you so much!

DarrenWong, 2023-01-19:
hey Nai, the sky-facing fisheye camera is not captured in the UrbanNav dataset. You can find the fisheye camera in our previous UrbanLoco dataset https://github.com/weisongwen/UrbanLoco

NAI-ZHU, 2023-01-24:
That's exactly what I'm looking for, thank you so much!

DarrenWong, 2023-01-25:
closed as resolved

closed 2023-01-25 by DarrenWong