weisongwen/UrbanNavDataset #24: imu axis

Lunar871, 2023-03-23:
Thank you for the open source dataset , may i know how is the IMU installed for the Tokyo dataset?

weisongwen, 2023-03-25:
@DarrenWong Hi darren, can you please kindly help with this? thanks

DarrenWong, 2023-03-25:
Hi, thank you for your interest in our dataset.

As [mentioned]( https://github.com/weisongwen/UrbanNavDataset#2-tokyo-dataset) in the readme that,
`Important Notes: the LiDAR calibration file for the LiDAR sensor, extrinsic parameters between sensors are not available now. If you wish to study the GNSS/LiDAR/IMU integration, we suggest using the dataset above collected in Hong Kong. However, the GNSS dataset from Tokyo is challenging which is collected in challenging urban canyons!
`
Suggest you use the Hong Kong dataset for the multi-sensor integration

closed 2023-04-07 by DarrenWong