weisongwen/UrbanNavDataset #25: Request to Upload a Paper PDF file

JokerJohn, 2023-07-02:
Hello, Welson. I deeply appreciate your remarkable work! I want to inquire whether uploading a PDF file of a research paper to this repository is possible. I am interested in studying this paper but have encountered difficulties downloading it. Neither ArXiv, ResearchGate, nor Sci-Hub has yielded any results. I would be immensely grateful if you could provide the PDF of the paper.

DarrenWong, 2023-07-03:
Hi, our journal paper with full details will be released soonly, please stay tuned!
You can have an introduction of our slides at the ION conference from here https://www.ion.org/publications/abstract.cfm?articleID=17895, or [slides](https://ba0b7fda-849d-46f9-ad0b-6a691cd3ab4a.usrfiles.com/ugd/ba0b7f_74d2d1bf25b44fd389a08dfc5b1e4dde.pdf) from our lab websites

JokerJohn, 2023-07-03:
> Hi, our journal paper with full details will be released soonly, please stay tuned! You can have an introduction of our slides at the ION conference from here [...]

Thank you for your prompt response. I know the publication venue for this research paper, but our university library does not have access to the copyrighted version of the paper from the ION conference. Could you provide a preprinted version of the paper? I am particularly interested in studying the hardware platform details discussed in the paper. Your assistance would be greatly appreciated.

DarrenWong, 2023-07-03:
Thanks again for your interest. Actually, we only present the slides at the ION conference, you can download them from  [here](https://ba0b7fda-849d-46f9-ad0b-6a691cd3ab4a.usrfiles.com/ugd/ba0b7f_74d2d1bf25b44fd389a08dfc5b1e4dde.pdf). We might not provide a preprinted version for the paper right now before the journal version published.

In terms of hardware platform, which kind of setup do you interested in? We can provide some details for you

JokerJohn, 2023-07-03:
> Thanks again for your interest. Actually, we only present the slides at the ION conference, you can download them from [...]

Ok, think you!

closed 2023-07-03 by JokerJohn

DarrenWong, 2023-07-07:
Hi you can refer to the latest publications from here: https://navi.ion.org/content/70/4/navi.602