weisongwen/UrbanNavDataset #28: Inquiry Regarding Labeled LOS/NLOS Dataset and CSV Format Data

lizajka, 2024-05-07:
Good day,

My name is Yelyzaveta Pervysheva, and I am currently working on multipath using AI technologies. I recently came across your dataset in RINEX format and found it to be quite intriguing for my research purposes.

I am particularly interested in obtaining a labeled dataset that distinguishes between Line-of-Sight (LOS) and Non-Line-of-Sight (NLOS) scenarios. Given the complexity of my research, having access to such a dataset would significantly contribute to the advancement of my work.

Additionally, I was wondering if it would be possible to acquire the same dataset in CSV format, since i struggle of convert those obs files.

Could you please let me know if such a labeled LOS/NLOS dataset is available? Furthermore, if the dataset can be provided in CSV format or if there are any conversion tools available, I would greatly appreciate your assistance in this matter.

Thank you very much for your time and consideration. You can contact me at yelyzaveta.pervysheva@tuni.fi

Warm regards,

Yelyzaveta Pervysheva

xtsysu1, 2024-05-08:
Subject: Inquiry Regarding Reference Velocity in UrbanNav-TK-20181219 Dataset

I would like to express my sincere gratitude for providing the UrbanNavDataset. It has been immensely beneficial to our research endeavors.

I am writing to ask for help regarding the reference Velocity data (Velocity X (m/s)  Velocity Y (m/s)  Velocity Z (m/s) in the reference.csv file) in the UrbanNav-TK-20181219 dataset, specifically for Shinjuku. During recent experiments conducted using the dataset, I noticed discrepancies in the reference Velocity values for Shinjuku. It appears there may be some conversion errors that I have failed to identify. Therefore, I am reaching out to you to inquire about any relevant information regarding the reference speed data for Shinjuku.

Any insights or guidance you could provide on this matter would be greatly appreciated.

Thank you very much for your attention to this inquiry. You can contact me at xiet67@mail2.sysu.edu.cn.

Best regards,
Ting Xie

ivannhf, 2024-05-09:
Hi Yelyzaveta, 

About the LOS and NLOS visibility labelling, only selected data is available at this moment, you can check with this link:
https://www.dropbox.com/scl/fi/sv412qfevakwfen8uky1a/20210521.medium-urban.whampoa.ublox.f9p_visibilityLabel.csv?rlkey=6jtwww04rxk4r2bwlu1wg5rhn&st=50x3ofz1&dl=0

For the CSV format of the data, the RINEX format is a standard format for different processing toolboxes to handle directly. Maybe some open-sourced toolboxes or libraries are a good starting point, such as:

1. pyrtklib (python): https://github.com/IPNL-POLYU/pyrtklib_demo5
2. GraphGNSSLib (C++): https://github.com/IPNL-POLYU/GraphGNSSLib
But surely, we can always consider publishing a CSV format that includes as many fields and information as possible to satisfy different fields' users.

Once again, thanks for your attention to our dataset. 

Regards,
Ivan

LizaTuni, 2024-08-01:
Thank you @ivannhf  so much for your reply . Can you please share with me any related to this datasets works and documentation how this dataset  collected  so I can cite it in my article

DarrenWong, 2024-08-20:
> Thank you @ivannhf so much for your reply . Can you please share with me any related to this datasets works and documentation how this dataset collected so I can cite it in my article

You can refer to the citation below,
Hsu, L. T., Wen, W., Chen, W., Liu, Z., Kubo, N., Suzuki, T., & Meguro, J. (2021). UrbanNav: An Open-Sourced Multisensory Dataset for Benchmarking Positioning Algorithms Designed for Urban Areas. In Proceedings of the 34th International Technical Meeting of the Satellite Division of the Institute of Navigation, ION GNSS+ 2021 (pp. 226-256). (Proceedings of the 34th International Technical Meeting of the Satellite Division of the Institute of Navigation, ION GNSS+ 2021). Institute of Navigation. https://doi.org/10.33012/2021.17895