weisongwen/UrbanNavDataset #31: imu参数问题

sumor-jiao, 2024-11-13:
您好，非常感谢您的开源分享。关于Tokyo Dataset UrbanNav-TK-20181219的学习过程中，我在您提供的Tamagawa-seiki TAG264文件中没有看到TAG264相关的信息。想咨询一下关于Tokyo Dataset所使用的imu的一些参数，例如随机游走、偏置标准差和相关时间。
 arw（ deg/sqrt(hr)）、  vrw（m/s/sqrt(hr)）、gbstd（ deg/hr）、abstd（mGal）、corrtime（hr） 
非常感谢！

DarrenWong, 2024-11-28:
你好，该款IMU的相关信息的确很少，你可以参考他们官网中的同类型产品，
https://mems.tamagawa-seiki.com/en/product/memsimu.html
[image]

或者可以参考有一篇GPS solutions论文中用到同一款的，他们matlab 代码是开源的，
Chen, K., Chang, G. & Chen, C. GINav: a MATLAB-based software for the data processing and analysis of a GNSS/INS integrated navigation system. GPS Solut 25, 108 (2021). https://doi.org/10.1007/s10291-021-01144-9

code: https://geodesy.noaa.gov/gps-toolbox/GINav.shtml

sumor-jiao, 2024-12-16:
非常感谢您的回复，这对我有很大的帮助！祝您工作顺利

closed 2025-01-15 by DarrenWong